jsk-ros-pkg/jsk_recognition #2781: jsk_recognition_msgs loaded catkin_virtualenv, but never invoked   'catkin_generate_virtualenv'

gyt290811, 2023-07-17:
hi, Do you meet this problem when compiling?
```
CMake Error at /opt/ros/noetic/share/catkin_virtualenv/cmake/catkin_install_python.cmake:30 (message):
  jsk_recognition_msgs loaded catkin_virtualenv, but never invoked
  'catkin_generate_virtualenv'
```
Or do you have any suggestions on this problem?
[image]

sktometometo, 2023-07-18:
`CMakeLists.txt` at current `master` does not contain `catkin_virtualenv` . https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_recognition_msgs/CMakeLists.txt

Could you show workspace configuration and `git status` result of `jsk_recognition_msgs`?

gyt290811, 2023-07-18:
> `CMakeLists.txt` at current `master` does not contain `catkin_virtualenv` . https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_recognition_msgs/CMakeLists.txt
> 
> Could you show workspace configuration and `git status` result of `jsk_recognition_msgs`?

This is the workspace configuration!
[image]

sktometometo, 2023-07-18:
Thanks!

I would like to see 1. whole workspace configuration and 2. current commit ID of `jsk_recognition` repository.

To export 1., Please install `python3-vcstool` and do `vcs export`

```bash
~/ros/ws_jsk_dxl_armed_turtlebot/src $ ls
dynamixel_motor  jsk_common  kobuki_desktop     turtlebot_apps          turtlebot_simulator
dynamixel_urdf   jsk_demos   robot-programming  turtlebot_interactions
jsk_3rdparty     kobuki      turtlebot          turtlebot_msgs
sktometometo@leopard
~/ros/ws_jsk_dxl_armed_turtlebot/src $ vcs export
repositories:
  dynamixel_motor:
    type: git
    url: https://github.com/arebgun/dynamixel_motor.git
    version: master
  dynamixel_urdf:
    type: git
    url: https://github.com/jsk-enshu/dynamixel_urdf.git
    version: master
  jsk_3rdparty:
    type: git
    url: https://github.com/sktometometo/jsk_3rdparty.git
    version: develop/turtlebot
  jsk_common:
    type: git
    url: https://github.com/sktometometo/jsk_common.git
    version: PR/update-interface-of-multi-map-server
  jsk_demos:
    type: git
    url: git@github.com:sktometometo/jsk_demos.git
    version: experiment/202303-elevator-operation
  kobuki:
    type: git
    url: https://github.com/yujinrobot/kobuki.git
    version: 4a8fdfc23906dbc4b8984ad8df408df1cb59bf85
  kobuki_desktop:
    type: git
    url: https://github.com/yujinrobot/kobuki_desktop.git
    version: 58459065087c7f9786cc2d1dc25a059280b5ca47
  robot-programming:
    type: git
    url: git@github.com:sktometometo/robot-programming.git
    version: develop
  turtlebot:
    type: git
    url: https://github.com/turtlebot/turtlebot.git
    version: feb176b10f13bcdc517e7f305c39327dac6388f0
  turtlebot_apps:
    type: git
    url: https://github.com/turtlebot/turtlebot_apps.git
    version: indigo
  turtlebot_interactions:
    type: git
    url: https://github.com/turtlebot/turtlebot_interactions.git
    version: 5e9d19c56e893e88af16690124c262976eec6fcd
  turtlebot_msgs:
    type: git
    url: https://github.com/turtlebot/turtlebot_msgs.git
    version: 834c448fb313ea7edb834d5552211daf53823395
  turtlebot_simulator:
    type: git
    url: https://github.com/turtlebot/turtlebot_simulator.git
    version: melodic

```

And For 2., could you please do `git show HEAD`?

```bash
~/ros/ws_jsk_dxl_armed_turtlebot/src/jsk_3rdparty $ git show HEAD
commit 185b836571fffb033907a0582e7e54700e8b9752 (HEAD -> develop/turtlebot, origin/develop/turtlebot)
Merge: 076f8432 5b9d9de2
Author: Koki Shinjo <sktometometo@gmail.com>
Date:   Wed Jul 27 15:56:43 2022 +0900

    Merge branch 'voicevox' into develop/turtlebot

```

gyt290811, 2023-07-18:
> Thanks!
> 
> I would like to see 1. whole workspace configuration and 2. current commit ID of `jsk_recognition` repository.
> 
> To export 1., Please install `python3-vcstool` and do `vcs export`
> 
> ```shell
> ~/ros/ws_jsk_dxl_armed_turtlebot/src $ ls
> dynamixel_motor  jsk_common  kobuki_desktop     turtlebot_apps          turtlebot_simulator
> dynamixel_urdf   jsk_demos   robot-programming  turtlebot_interactions
> jsk_3rdparty     kobuki      turtlebot          turtlebot_msgs
> sktometometo@leopard
> ~/ros/ws_jsk_dxl_armed_turtlebot/src $ vcs export
> repositories:
>   dynamixel_motor:
>     type: git
>     url: https://github.com/arebgun/dynamixel_motor.git
>     version: master
>   dynamixel_urdf:
>     type: git
>     url: https://github.com/jsk-enshu/dynamixel_urdf.git
>     version: master
>   jsk_3rdparty:
>     type: git
>     url: https://github.com/sktometometo/jsk_3rdparty.git
>     version: develop/turtlebot
>   jsk_common:
>     type: git
>     url: https://github.com/sktometometo/jsk_common.git
>     version: PR/update-interface-of-multi-map-server
>   jsk_demos:
>     type: git
>     url: git@github.com:sktometometo/jsk_demos.git
>     version: experiment/202303-elevator-operation
>   kobuki:
>     type: git
>     url: https://github.com/yujinrobot/kobuki.git
>     version: 4a8fdfc23906dbc4b8984ad8df408df1cb59bf85
>   kobuki_desktop:
>     type: git
>     url: https://github.com/yujinrobot/kobuki_desktop.git
>     version: 58459065087c7f9786cc2d1dc25a059280b5ca47
>   robot-programming:
>     type: git
>     url: git@github.com:sktometometo/robot-programming.git
>     version: develop
>   turtlebot:
>     type: git
>     url: https://github.com/turtlebot/turtlebot.git
>     version: feb176b10f13bcdc517e7f305c39327dac6388f0
>   turtlebot_apps:
>     type: git
>     url: https://github.com/turtlebot/turtlebot_apps.git
>     version: indigo
>   turtlebot_interactions:
>     type: git
>     url: https://github.com/turtlebot/turtlebot_interactions.git
>     version: 5e9d19c56e893e88af16690124c262976eec6fcd
>   turtlebot_msgs:
>     type: git
>     url: https://github.com/turtlebot/turtlebot_msgs.git
>     version: 834c448fb313ea7edb834d5552211daf53823395
>   turtlebot_simulator:
>     type: git
>     url: https://github.com/turtlebot/turtlebot_simulator.git
>     version: melodic
> ```
> 
> And For 2., could you please do `git show HEAD`?
> 
> ```shell
> ~/ros/ws_jsk_dxl_armed_turtlebot/src/jsk_3rdparty $ git show HEAD
> commit 185b836571fffb033907a0582e7e54700e8b9752 (HEAD -> develop/turtlebot, origin/develop/turtlebot)
> Merge: 076f8432 5b9d9de2
> Author: Koki Shinjo <sktometometo@gmail.com>
> Date:   Wed Jul 27 15:56:43 2022 +0900
> 
>     Merge branch 'voicevox' into develop/turtlebot
> ```

Thanks very much!
**1. whole workspace configuration**
```
yc@yc-NUC11PHi7:~/catkin_ws/src$ vcs export
repositories:
  jsk_recognition:
    type: git
    url: https://github.com/jsk-ros-pkg/jsk_recognition.git
    version: master
yc@yc-NUC11PHi7:~/catkin_ws/src$ 
```
**2. current commit ID**
```
yc@yc-NUC11PHi7:~/catkin_ws/src/jsk_recognition$ git show HEAD
commit f264687999780859162a8ac19e84c96f92c14627 (HEAD -> master, origin/master, origin/HEAD)
Merge: ef04a200 6200b42d
Author: Kei Okada <k-okada@jsk.t.u-tokyo.ac.jp>
Date:   Mon Jun 26 14:18:30 2023 +0900

    Merge pull request #2774 from mqcmd196/PR/pcl_nodelet_log
    
    [jsk_pcl_ros_utils] use nodelet logger and fix logging info

yc@yc-NUC11PHi7:~/catkin_ws/src/jsk_recognition$ 

```

sktometometo, 2023-07-18:
Hmm... that error does not occur in my environment. (Ubuntu 20.04 + noetic)

```bash
~/ros/ws_test/src $ catkin config
-------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/noetic
Workspace:                   /home/sktometometo/ros/ws_test
-------------------------------------------------------------------
Build Space:        [exists] /home/sktometometo/ros/ws_test/build
Devel Space:        [exists] /home/sktometometo/ros/ws_test/devel
Install Space:      [unused] /home/sktometometo/ros/ws_test/install
Log Space:          [exists] /home/sktometometo/ros/ws_test/logs
Source Space:       [exists] /home/sktometometo/ros/ws_test/src
DESTDIR:            [unused] None
-------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------------------
Buildlisted Packages:        None
Skiplisted Packages:         None
-------------------------------------------------------------------
Workspace configuration appears valid.
-------------------------------------------------------------------
sktometometo@lee
[gitbranch:]
[ROS VERSION:1]
[ROS workspace:]
[07:34 午前 水  7月 19]
~/ros/ws_test/src $ catkin cl^C
sktometometo@lee
[gitbranch:]
[ROS VERSION:1]
[ROS workspace:]
[07:34 午前 水  7月 19]
~/ros/ws_test/src $ vcs export
repositories:
  jsk_recognition:
    type: git
    url: git@github.com:jsk-ros-pkg/jsk_recognition.git
    version: master
sktometometo@lee
[gitbranch:]
[ROS VERSION:1]
[ROS workspace:]
[07:34 午前 水  7月 19]
~/ros/ws_test/src $ cd jsk_recognition/
sktometometo@lee
[gitbranch:master]
[ROS VERSION:1]
[ROS workspace:]
[07:34 午前 水  7月 19]
~/ros/ws_test/src/jsk_recognition $ git show HEAD
commit f264687999780859162a8ac19e84c96f92c14627 (HEAD -> master, origin/master, origin/HEAD)
Merge: ef04a200 6200b42d
Author: Kei Okada <k-okada@jsk.t.u-tokyo.ac.jp>
Date:   Mon Jun 26 14:18:30 2023 +0900

    Merge pull request #2774 from mqcmd196/PR/pcl_nodelet_log
    
    [jsk_pcl_ros_utils] use nodelet logger and fix logging info

sktometometo@lee
[gitbranch:master]
[... 35 lines cut ...]
-------------------------------------------------------------------
[build] Found 11 packages in 0.0 seconds.                                                                                                                                                                         
[build] Package table is up to date.                                                                                                                                                                              
Starting  >>> jsk_recognition_msgs                                                                                                                                                                                
Finished  <<< jsk_recognition_msgs                [ 2.4 seconds ]                                                                                                                                                 
[build] Summary: All 1 packages succeeded!                                                                                                                                                                        
[build]   Ignored:   10 packages were skipped or are skiplisted.                                                                                                                                                  
[build]   Warnings:  None.                                                                                                                                                                                        
[build]   Abandoned: None.                                                                                                                                                                                        
[build]   Failed:    None.                                                                                                                                                                                        
[build] Runtime: 2.5 seconds total.                             
```

Could you create another workspace, download repo, install dependency and build it again?

```bash
sudo apt update && sudo apt upgrade
mkdir -p ~/catkin_test_ws/src
cd catkin_test_ws
catkin init
cd src
git clone https://github.com/jsk-ros-pkg/jsk_recognition
rosdep install --from-paths . --ignore-src -y -r
catkin build jsk_recognition_msgs
```

mqcmd196, 2023-07-19:
@gyt290811 
BTW, what build system are you using? [Official](https://github.com/locusrobotics/catkin_virtualenv) says 

> catkin-tools, colcon, catkin_make_isolated (i.e. ros_buildfarm), but not catkin_make

It seems that you use `catkin_make` ?  If so, maybe `catkin_make` wrongly invokes `catkin_install_python` even if it is not written in the current package's cmake.

gyt290811, 2023-07-19:
@mqcmd196 
Yeah, I just realized this issue and  the command I used to compile the workspace was indeed `catkin make`

gyt290811, 2023-07-19:
@sktometometo 
I compiled `jsk_recognition_msgs` separately following your suggestion, and the results are the same as yours. 
Meanwhile, I realized that the command I used to compile the workspace was  `catkin make`. Maybe it's because of the way I compiled the repo in workspace!
So, how did you compile this repo? 
I didn't find the steps in `install_insta360_air.md `

sktometometo, 2023-07-19:
We use `catkin build` by default for building catkin packages. I think `catkin_make` command is deprecated now and `catkin_make_isolated` is used by default now.
See https://answers.ros.org/question/320613/catkin_make-vs-catkin_make_isolated-which-is-preferred/ for details.

closed 2023-07-21 by sktometometo