jsk-ros-pkg/jsk_recognition #2807: Sample data cannot be downloaded.

sktometometo, 2023-11-16:
Some sample data are not downloaded during `catkin build`.

And this seems to come from https://github.com/wkentaro/gdown/issues/43

We have to find some workarounds about this problem (that we cannot download file which is downloaded many times) or think about another solution for storages about sample data.

```bash
	Cannot retrieve the public link of the file. You may need to change
	the permission to 'Anyone with the link', or have had many accesses. 

You may still be able to access the file from the browser:

	 https://drive.google.com/uc?id=1B1oC6C9ZrjOQTScmbWwR_Y91GAFIg4bu 

 	Cannot retrieve the public link of the file. You may need to change
	the permission to 'Anyone with the link', or have had many accesses. 

You may still be able to access the file from the browser:

	 https://drive.google.com/uc?id=1B1oC6C9ZrjOQTScmbWwR_Y91GAFIg4bu 

[/home/sktometometo/.ros/data/jsk_perception/2020-11-11-15-58-47-d435-floor.bag] could not download file. maximum download trial exceeded.

 	Cannot retrieve the public link of the file. You may need to change
	the permission to 'Anyone with the link', or have had many accesses. 

You may still be able to access the file from the browser:

	 https://drive.google.com/uc?id=1f99gIK92EsGvVI5ahHAV5g8xoEHyQhtv 

 	Cannot retrieve the public link of the file. You may need to change
	the permission to 'Anyone with the link', or have had many accesses. 

You may still be able to access the file from the browser:

	 https://drive.google.com/uc?id=1f99gIK92EsGvVI5ahHAV5g8xoEHyQhtv 

[/home/sktometometo/.ros/data/jsk_perception/insta360_air.bag.tgz] could not download file. maximum download trial exceeded.

 	Cannot retrieve the public link of the file. You may need to change
	the permission to 'Anyone with the link', or have had many accesses. 

You may still be able to access the file from the browser:

	 https://drive.google.com/uc?id=1WKnd-gDgp3GCWOAPtwg1lkrPqBiIyB_l 

 	Cannot retrieve the public link of the file. You may need to change
	the permission to 'Anyone with the link', or have had many accesses. 

You may still be able to access the file from the browser:

	 https://drive.google.com/uc?id=1WKnd-gDgp3GCWOAPtwg1lkrPqBiIyB_l 

[/home/sktometometo/.ros/data/jsk_perception/fisheye_stitcher_grid_xd_yd_3840x1920_fetch15.yml.gz] could not download file. maximum download trial exceeded.

 	Cannot retrieve the public link of the file. You may need to change
	the permission to 'Anyone with the link', or have had many accesses. 

You may still be able to access the file from the browser:

	 https://drive.google.com/uc?id=1SZwyUT14uEDuKvBzv6K0iOcYJoIIbaI0 

 	Cannot retrieve the public link of the file. You may need to change
	the permission to 'Anyone with the link', or have had many accesses. 

You may still be able to access the file from the browser:

	 https://drive.google.com/uc?id=1SZwyUT14uEDuKvBzv6K0iOcYJoIIbaI0 

[/home/sktometometo/.ros/data/jsk_perception/fisheye_stitcher_grid_xd_yd_3840x1920_fetch1075.yml.gz] could not download file. maximum download trial exceeded.
[... 170 lines cut ...]
You may still be able to access the file from the browser:

	 https://drive.google.com/uc?id=1LhuGYlPNXEJW-G3dOwxD60z3k9KvYRxg 

[/home/sktometometo/.ros/data/jsk_pcl_ros_utils/2017-02-05-16-11-09_shelf_bin.bag] could not download file. maximum download trial exceeded.

 	Cannot retrieve the public link of the file. You may need to change
	the permission to 'Anyone with the link', or have had many accesses. 

You may still be able to access the file from the browser:

	 https://drive.google.com/uc?id=1thJxY4BJrU1ltxBoWstTaMEQfhFmsls4 

[/home/sktometometo/.ros/data/jsk_pcl_ros_utils/arc2017_4objects.bag] could not download file. maximum download trial exceeded.

 	Cannot retrieve the public link of the file. You may need to change
	the permission to 'Anyone with the link', or have had many accesses. 

You may still be able to access the file from the browser:

	 https://drive.google.com/uc?id=1RvpUWnPJogYw_n7IRfmt6ONpUD1ZETus 

[/home/sktometometo/.ros/data/jsk_pcl_ros_utils/2019-11-03-hsr-multi-objects-in-shelf.bag] could not download file. maximum download trial exceeded.

```

k-okada, 2023-11-16:
Are you saying that  the error is caused by  google's bandwidth limitation? if so, how about preparing a  file packed with sample data?  For exampe, downloading https://drive.google.com/drive/u/1/folders/1uVlR00jbwYHgXPIfQ-yLsZN5d5yrjxpw and extracting it to your home direfctory? If there are missing sample data, please add that fie into tgz.

sktometometo, 2023-11-20:
> Are you saying that the error is caused by google's bandwidth limitation?

The problems, according to https://github.com/wkentaro/gdown/issues/43#issuecomment-1382813663, there are some limitations for access without OAuth to Google Drive and `gdown` cannot resolve it.

> if so, how about preparing a file packed with sample data?

This may not fundamentally resolve the problem because the same problem will happen if new file will be downloaded many times.

With current configuration, sample data will be downloaded automatically if it is built with `catkin build`. So I am suspecting there are some people in the world who run CI where `jsk_recognition` is downloaded, built and sample data is downloaded and download capacity is ran out.

If so we also need default configuration about `catkin build` of jsk_recognition so that sample data is not downloaded automatically. How do you think? @k-okada