jsk-ros-pkg/jsk_recognition #2827: Build error in Install PCL from source

Kanazawanaoaki, 2024-04-21:
I tried to do "Install PCL from source with CUDA" on https://jsk-docs.readthedocs.io/projects/jsk_recognition/en/latest/install_pcl_from_source.html to use Kinfu, but errors occurred.

### Error pattern 1
```
/home/kanazawa/pcl-pcl-1.8.1/segmentation/include/pcl/segmentation/plane_coefficient_comparator.h:144:17: error: invalid initialization of reference of type 'const std::vector<float>&' from expression of type 'const boost::shared_ptr<std::vector<float> >'
  144 |         return (plane_coeff_d_);
      |                ~^~~~~~~~~~~~~~~
```
Related: https://github.com/PointCloudLibrary/pcl/issues/2282

### Error pattern 2
```
/home/kanazawa/pcl-pcl-1.8.1/visualization/src/pcl_visualizer.cpp:103:10: fatal error: boost/uuid/sha1.hpp: No such file or directory
  103 | #include <boost/uuid/sha1.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```
Related: https://github.com/PointCloudLibrary/pcl/issues/2682

### Solution
These problems seem to have been resolved in PCL version 1.9.0, so I was able to build as follows.
```
cd
wget https://github.com/PointCloudLibrary/pcl/archive/pcl-1.9.0.tar.gz
tar xvzf pcl-1.9.0.tar.gz
cd pcl-pcl-1.9.0
mkdir build
cd build
cmake .. -DCMAKE_BUILD_TYPE=Release -DWITH_CUDA:BOOL=ON
make -j 4
sudo make install
````
If kinfu is successfully executed and there are no problems, I will create a PR.

mqcmd196, 2024-04-21:
> Related: PointCloudLibrary/pcl#2282
> Related: PointCloudLibrary/pcl#2682

I think the version would depend on the current Ubuntu version.

If you use Ubuntu 20.04, you might download the PCL 1.10.0 source.
Canonical releases the PCL by apt with 1.10.0
https://packages.ubuntu.com/source/focal/pcl .

Installing different versions of packages not released by apt will cause
ABI incompatibility and segmentation faults in libraries and executables
that depend on PCL.

knorth55, 2024-04-22:
When we want to build jsk_pcl_ros with your own `PCL`, you need to carefully check PCL version match.
`ldd libjsk_pcl_ros.so | grep pcl`
At least, we need to build these packages when we use PCL 1.8 on indigo or kinetic, i remember.

cc. @pazeshun 
https://github.com/start-jsk/jsk_apc/blob/a468d3e44b3964a69faa372d9fe141b72ab13f94/fc.rosinstall.kinetic#L21-L32