jsk-ros-pkg/jsk_recognition #2828: Kinfuを実行しようとするとFailed to load nodelet [/kinfu]になる

Kanazawanaoaki, 2024-04-21:
https://github.com/jsk-ros-pkg/jsk_recognition/issues/2827 に示すようにInstall PCL from source with CUDAの手順を行い，jsk_recognitionをbuildした後，
```
roslaunch jsk_pcl_ros sample_kinfu.launch
```
のサンプルを実行すると以下のエラーが発生してkinfuのプロセスが死んでしまいました．
```
process[kinfu-24]: started with pid [1121253]
type is jsk_pcl/Kinfu
[ERROR] [1713681488.335802289]: Skipped loading plugin with error: XML Document '/opt/ros/noetic/share/prosilica_camera/plugins/nodelet_plugins.xml' has no Root Element. This likely means the XML is malformed or missing..
[ERROR] [1713681488.398246717]: Skipped loading plugin with error: XML Document '/opt/ros/noetic/share/prosilica_camera/plugins/nodelet_plugins.xml' has no Root Element. This likely means the XML is malformed or missing..
[ERROR] [1713681488.712618028, 1498131092.821610080]: Skipped loading plugin with error: XML Document '/opt/ros/noetic/share/prosilica_camera/plugins/nodelet_plugins.xml' has no Root Element. This likely means the XML is malformed or missing..
[ERROR] [1713681488.714973268, 1498131092.821610080]: Failed to load nodelet [/kinfu] of type [jsk_pcl/Kinfu] even after refreshing the cache: MultiLibraryClassLoader: Could not create object of class type jsk_pcl_ros::Kinfu as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[ERROR] [1713681488.714992792, 1498131092.821610080]: The error before refreshing the cache was: MultiLibraryClassLoader: Could not create object of class type jsk_pcl_ros::Kinfu as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[kinfu-24] process has died [pid 1121253, exit code 255, cmd /opt/ros/noetic/lib/nodelet/nodelet standalone jsk_pcl/Kinfu ~input/camera_info:=/camera/depth_registered/camera_info ~input/depth:=/camera/depth_registered/hw_registered/image_rect_raw ~input/color:=/camera/rgb/image_rect_color __name:=kinfu __log:=/home/kanazawa/.ros/log/b66322fc-ffa9-11ee-a86a-79b99c0bb4cb/kinfu-24.log].
log file: /home/kanazawa/.ros/log/b66322fc-ffa9-11ee-a86a-79b99c0bb4cb/kinfu-24*.log
```
エラーメッセージ的にはhttps://github.com/jsk-ros-pkg/jsk_recognition/issues/1482 や https://sourceforge.net/p/jsk-ros-pkg/tickets/119/ が近そうですがよくわかりません．

https://answers.ros.org/question/110316/pluginception-build-factory/
を見るとライブラリのリンクが関係しそうなので，build周りを確認していると，
build時に`PCL_GPU_KINFU_LARGE_SCALE_FOUND`の変数がtrue?になっておらず
https://github.com/jsk-ros-pkg/jsk_recognition/blob/908748677833a22832c1934e367ef32e4b79912e/jsk_pcl_ros/CMakeLists.txt#L422-L426
辺りの処理が行われていないのが原因のように思います．　　
この変数やPCL_GPU_KINFU_LARGE_SCALE_LIBRARYやPCL_GPU_CONTAINERS_LIBRAR辺りの変数はbuild時に手動で指定する必要があるのでしょうか？　　
あるいはインストールの手順が間違っているor不足しているのでしょうか？

mqcmd196, 2024-04-21:
https://github.com/jsk-ros-pkg/jsk_recognition/blob/908748677833a22832c1934e367ef32e4b79912e/jsk_pcl_ros/CMakeLists.txt#L66C35-L66C56

ここでセットされるはずです．`QUIET` を消して，CMakeの出力を教えてもらえますか？

Kanazawanaoaki, 2024-04-21:
ありがとう，とりあえず実行したlogを貼っておきます．
https://gist.github.com/Kanazawanaoaki/2bd923abbd818b9470805bde0b8df12e#file-kinfu_debug_catkin_log_20240421-log

その部分の後ろに
```cmake
# Function to check if a variable is defined and print its value if defined
function(print_if_defined variable_name)
    if(DEFINED ${variable_name})
        message("${variable_name} is defined and its value is ${${variable_name}}.")
    else()
        message("${variable_name} is not defined.")
    endif()
endfunction()

print_if_defined(PCL_GPU_KINFU_LARGE_SCALE_FOUND)
print_if_defined(PCL_GPU_KINFU_LARGE_SCALE_LIBRARY)
print_if_defined(PCL_GPU_CONTAINERS_LIBRARY)
```
を追加したら
```
PCL_GPU_KINFU_LARGE_SCALE_FOUND is defined and its value is FALSE.
PCL_GPU_KINFU_LARGE_SCALE_LIBRARY is defined and its value is PCL_GPU_KINFU_LARGE_SCALE_LIBRARY-NOTFOUND.
PCL_GPU_CONTAINERS_LIBRARY is not defined.
```
になっていたので，セットはされているけど無しでセットされているみたいですね．

mqcmd196, 2024-04-21:
なるほど，ということはPCLのソースビルドが出来ていないか，ビルドしたがそこに適切にPATHが通っていない可能性が高いです．
ただ#2827 を見る感じ，1.9.0でビルドそのものは出来ているっぽいかな

あとCMakeの出力が見たかったので catkin bt --verbose の結果がほしかったです．

とりあえず送ってくれたログを見ると
https://gist.github.com/Kanazawanaoaki/2bd923abbd818b9470805bde0b8df12e#file-kinfu_debug_catkin_log_20240421-log-L737

で `/usr/include/pcl-1.10`
を見にいっていて，ソースビルドしたPCLを見にいっていない気がするので，シェルを立ち上げ直すなり，CMAKE_PREFIX_PATHを新たに設定するなりする必要があると思いますが，/usr/local
以下は標準で見る気がする

`catkin clean jsk_pcl_ros` して
`catkin build --force-cmake jsk_pcl_ros --verbose`

のログを見せてください

Kanazawanaoaki, 2024-04-21:
すみません，なるほどです．

シェルを立ち上げ直して，実行しました．
長すぎて[途中からのログ](https://gist.github.com/Kanazawanaoaki/2bd923abbd818b9470805bde0b8df12e#file-kinfu_debug_catkin_verbose_log_20240421-log　)と，[実行し直した際の途中のログ](https://gist.github.com/Kanazawanaoaki/84c04b128b68c8c45616b795a2457702#file-kinfu_debug_catkin_verbose_log_v2_20240421-log)のみですが，
https://gist.github.com/Kanazawanaoaki/84c04b128b68c8c45616b795a2457702#file-kinfu_debug_catkin_verbose_log_v2_20240421-log-L19
```bash
CMake Deprecation Warning at /usr/local/share/pcl-1.9/PCLConfig.cmake:42 (cmake_policy):
```
で1.9の方を見ていそうなのと，
https://gist.github.com/Kanazawanaoaki/84c04b128b68c8c45616b795a2457702#file-kinfu_debug_catkin_verbose_log_v2_20240421-log-L31-L48
の辺りで，
```
-- Could NOT find PCL_GPU_KINFU_LARGE_SCALE (missing: PCL_GPU_KINFU_LARGE_SCALE_LIBRARY PCL_GPU_KINFU_LARGE_SCALE_INCLUDE_DIR)
```
でPCL_GPU_KINFU_LARGE_SCALEは見つけられてなさそうです．

mqcmd196, 2024-04-21:
ごめん，序盤のログが一番見たかったです．
workspaceをsourceして

```
roscd jsk_pcl_ros
mkdir build && cd build
cmake ..
```
の実行結果を教えてください

mqcmd196, 2024-04-21:
> -- Could NOT find PCL_GPU_KINFU_LARGE_SCALE (missing:
PCL_GPU_KINFU_LARGE_SCALE_LIBRARY PCL_GPU_KINFU_LARGE_SCALE_INCLUDE_DIR)

たしかに見つかってなさそうですね

Kanazawanaoaki, 2024-04-21:
https://github.com/jsk-ros-pkg/jsk_recognition/issues/2827#issuecomment-2067988063
を参考に，1.10.0のものでbuildし直して実行したログです．
https://gist.github.com/Kanazawanaoaki/d2ef0974827c5bbde843f52d4a1276c6#file-20240421_1947_cmake_log-log

```
-- Could NOT find PCL_GPU_KINFU_LARGE_SCALE (missing: PCL_GPU_KINFU_LARGE_SCALE_LIBRARY PCL_GPU_KINFU_LARGE_SCALE_INCLUDE_DIR) 
```
でやはり見つかっていなさそうです．．．

mqcmd196, 2024-04-21:
PCLのビルドのときにCUDAアリでビルドされているか見てほしいのですが，
```
include("${PCL_SOURCE_DIR}/cmake/pcl_find_cuda.cmake")
```
`-DCUDA=ON` とかすると https://github.com/PointCloudLibrary/pcl/blob/pcl-1.10.0/CMakeLists.txt#L366 が呼ばれることになっていて，
https://github.com/PointCloudLibrary/pcl/blob/pcl-1.10.0/cmake/pcl_find_cuda.cmake を見ると，`-DCUDA=ON` だが，CUDAが見つからないときも，特にエラーでコケることなく，そのままビルドが続行されていそうですね．

なのでPCLをビルドするときのCMakeコマンドのログをください

Kanazawanaoaki, 2024-04-22:
遅くなりすみません．
これは今実行したログですが，
https://gist.github.com/Kanazawanaoaki/6d2237067d5986516c8317407c78514a#file-pcl_cmake_log_20240422-log
最初に1.10.0のPCLをbuildしたときも以下のようなログが出ていたのでCUDAは見つかっているとは思います．
```
-- Found CUDA Toolkit v10.1
```

Kanazawanaoaki, 2024-04-22:
https://github.com/PointCloudLibrary/pcl/issues/466#issuecomment-33262303
などのログと比較すると，
https://gist.github.com/Kanazawanaoaki/6d2237067d5986516c8317407c78514a#file-pcl_cmake_log_20240422-log-L181-L208
ここに
```
--   gpu_kinfu
--   gpu_kinfu_large_scale
```
とかが無いのが問題のようで，やはりPCLのBuildでうまくいっていないみたいです．

Kanazawanaoaki, 2024-04-22:
その後
```
cd
wget https://github.com/PointCloudLibrary/pcl/archive/pcl-1.10.0.tar.gz
tar xvzf pcl-1.10.0.tar.gz
cd pcl-pcl-1.10.0
mkdir build
cd build
cmake .. -DCMAKE_BUILD_TYPE=Release -DWITH_CUDA:BOOL=ON -DBUILD_GPU=ON
make -j 4
sudo make install
```
のmake -j 4の手順で
```
#error -- unsupported GNU version! gcc versions later than 8 are not supported!
```
のエラーが出たので，https://qiita.com/PinappleHunter/items/d07ee6f96cf60959b635 , https://linuxize.com/post/how-to-install-gcc-on-ubuntu-20-04/#installing-multiple-gcc-versions
を参考にgcc 8を使ってcmakeとmakeを実行．

```
In file included from /home/kanazawa/pcl-pcl-1.10.0/gpu/kinfu/tools/evaluation.cpp:37:
/home/kanazawa/pcl-pcl-1.10.0/gpu/kinfu/tools/evaluation.h:53:15: error: ‘shared_ptr’ does not name a type; did you mean ‘char16_t’?
   using Ptr = shared_ptr<Evaluation>;
               ^~~~~~~~~~
               char16_t
```
のエラーが出てmakeできなくなりました．
https://github.com/PointCloudLibrary/pcl/issues/5836 によると，`sudo apt remove libpcl*`で解決するらしいが，そうすると大量のpcl依存のパッケージがアンインストールされるので他の回避策を模索中です．