jsk-ros-pkg/jsk_recognition #2840: Polygon Array not showing Polygon when only border is False

mitch-galea, 2024-05-29:
I have noticed that some polygons are not showing when `only border` is False.
These polygons are usually concave.

I have a python script below with the polygon. I am publishing a single polygon as both a Polygon and PolygonArray. The Polygon works fine but the PolygonArray does not with when `only border` is False.

[image]
[image]

```
#!/usr/bin/env python

import rospy
import time

from geometry_msgs.msg import PolygonStamped, Polygon, Point32
from jsk_recognition_msgs.msg import PolygonArray

class Publisher:

    def __init__(self) -> None:
        """Initialises the node"""

        rospy.init_node("publish_polygon")

        # wait for time to init
        while rospy.get_rostime().to_sec() == 0.0:
            time.sleep(0.1)

        points = [
            {"x": 845.4697875976562, "y": -1329.1895751953125, "z": 0.0},
            {"x": 110.03823852539062, "y": -1324.4365234375, "z": 0.0},
            {"x": 109.90898132324219, "y": -1344.43603515625, "z": 0.0},
            {"x": 846.8936157226562, "y": -1349.1990966796875, "z": 0.0},
            {"x": 847.5587768554688, "y": -1349.1812744140625, "z": 0.0},
            {"x": 848.2213134765625, "y": -1349.1192626953125, "z": 0.0},
            {"x": 848.7095336914062, "y": -1349.05712890625, "z": 0.0},
            {"x": 849.2949829101562, "y": -1348.96484375, "z": 0.0},
            {"x": 849.8739013671875, "y": -1348.8380126953125, "z": 0.0},
            {"x": 850.4443359375, "y": -1348.67724609375, "z": 0.0},
            {"x": 851.0042724609375, "y": -1348.48291015625, "z": 0.0},
            {"x": 1158.0098876953125, "y": -1231.6077880859375, "z": 0.0},
            {"x": 1150.8941650390625, "y": -1212.9163818359375, "z": 0.0},
        ]

        header = rospy.Header(stamp=rospy.Time.now(), frame_id="init_fix_accurate_link")

        polygon = Polygon(points=[Point32(**point) for point in points])
        polygon_stamped = PolygonStamped(
            header=header,
            polygon=polygon,
        )
        polygon_array = PolygonArray(header=header, polygons=[polygon_stamped], labels=[0], likelihood=[1.0])

        self.polygon_publisher = rospy.Publisher(
            name="polygon", data_class=PolygonStamped, latch=True, queue_size=1
        )

        self.polygons_publisher = rospy.Publisher(
            name="polygon_array", data_class=PolygonArray, latch=True, queue_size=1
        )

        self.polygon_publisher.publish(polygon_stamped)
        self.polygons_publisher.publish(polygon_array)

    def run(self) -> None:
        """main run loop"""
        rospy.spin()

def main() -> None:
    """main function"""
    node = Publisher()
    node.run()

if __name__ == "__main__":
    main()

```

iory, 2024-05-29:
Thanks for your reports.
The cause of this issue is that the displayed polygon is too large, making the line width appear relatively small and seemingly invisible. The following program resizes the polygon to a more realistic size, and the polygon is now displayed.

Why is it being rendered at this size?

```
#!/usr/bin/env python

import rospy
import time

from geometry_msgs.msg import PolygonStamped, Polygon, Point32
from jsk_recognition_msgs.msg import PolygonArray

class Publisher:

    def __init__(self) -> None:
        """Initialises the node"""

        rospy.init_node("publish_polygon")

        # wait for time to init
        while rospy.get_rostime().to_sec() == 0.0:
            time.sleep(0.1)

        points = [
            {"x": 0.001 * 845.4697875976562, "y": 0.001 * -1329.1895751953125, "z": 0.0},
            {"x": 0.001 * 110.03823852539062, "y": 0.001 * -1324.4365234375, "z": 0.0},
            {"x": 0.001 * 109.90898132324219, "y": 0.001 * -1344.43603515625, "z": 0.0},
            {"x": 0.001 * 846.8936157226562, "y": 0.001 * -1349.1990966796875, "z": 0.0},
            {"x": 0.001 * 847.5587768554688, "y": 0.001 * -1349.1812744140625, "z": 0.0},
            {"x": 0.001 * 848.2213134765625, "y": 0.001 * -1349.1192626953125, "z": 0.0},
            {"x": 0.001 * 848.7095336914062, "y": 0.001 * -1349.05712890625, "z": 0.0},
            {"x": 0.001 * 849.2949829101562, "y": 0.001 * -1348.96484375, "z": 0.0},
            {"x": 0.001 * 849.8739013671875, "y": 0.001 * -1348.8380126953125, "z": 0.0},
            {"x": 0.001 * 850.4443359375, "y": 0.001 * -1348.67724609375, "z": 0.0},
            {"x": 0.001 * 851.0042724609375, "y": 0.001 * -1348.48291015625, "z": 0.0},
            {"x": 0.001 * 1158.0098876953125, "y": 0.001 * -1231.6077880859375, "z": 0.0},
            {"x": 0.001 * 1150.8941650390625, "y": 0.001 * -1212.9163818359375, "z": 0.0},
        ]

        header = rospy.Header(stamp=rospy.Time.now(), frame_id="init_fix_accurate_link")

        polygon = Polygon(points=[Point32(**point) for point in points])
        polygon_stamped = PolygonStamped(
            header=header,
            polygon=polygon,
        )
        polygon_array = PolygonArray(header=header, polygons=[polygon_stamped], labels=[0], likelihood=[1.0])

        self.polygon_publisher = rospy.Publisher(
            name="polygon", data_class=PolygonStamped, latch=True, queue_size=1
        )

        self.polygons_publisher = rospy.Publisher(
            name="polygon_array", data_class=PolygonArray, latch=True, queue_size=1
        )

        self.polygon_publisher.publish(polygon_stamped)
        self.polygons_publisher.publish(polygon_array)

    def run(self) -> None:
        """main run loop"""
        rospy.spin()

def main() -> None:
    """main function"""
    node = Publisher()
    node.run()

if __name__ == "__main__":
    main()
```
[image]

mitch-galea, 2024-05-30:
I can see the polygon, albiet very thinly when `only border` is ticked and I understand that this is due to the size. My issue is that when `only border` is unticked (desire is a filled polygon) then the polygon does not show up

mitch-galea, 2024-05-30:
just checked with your script and yeah looks like they show up filled, when reduced in size. This only seems to happen with thin concave polygons.

iory, 2024-05-30:
I see.
I fixed the problem. Please try it!
https://github.com/jsk-ros-pkg/jsk_recognition/pull/2841

mitch-galea, 2025-08-13:
I have built jsk_recognition_utils and jsk_rviz_plugins with the fix and it didn't fix the problem