jsk-ros-pkg/jsk_recognition #2849: [jsk_perception/vqa_node] No module named 'jsk_perception.vil_inference_client'

kochigami, 2024-11-05:
Hi, 
I’d like to use VQA (https://github.com/jsk-ros-pkg/jsk_recognition/pull/2730).

Using ROS noetic, I followed https://jsk-docs.readthedocs.io/projects/jsk_recognition/en/latest/jsk_perception/nodes/vqa_node.html#prerequisite, https://jsk-docs.readthedocs.io/projects/jsk_recognition/en/latest/jsk_perception/nodes/vqa_node.html#build-the-docker-image,
and executed `roscd jsk_perception` & `catkin bt`.

And then, I tried the following commands:

<terminal 1>
```
roscore
```

<terminal 2>
```
rosrun usb_cam usb_cam_node
```

<terminal 3>
```
roslaunch jsk_perception  vqa.launch run_api:=true VQA_INPUT_IMAGE:=/usb_cam/image_raw gui:=true
```

On terminal 3, I got the following error:
```
Traceback (most recent call last):
  File "/home/kochigami/catkin_ws/src/jsk_recognition/jsk_perception/node_scripts/vqa_node.py", line 4, in <module>
    from jsk_perception.vil_inference_client import OFAClientNode
ModuleNotFoundError: No module named 'jsk_perception.vil_inference_client'
```
Whole error message:
```
kochigami@kochigami-ThinkPad-P16-Gen-1:~/catkin_ws/src/jsk_recognition/jsk_perception/docker$ roslaunch jsk_perception  vqa.launch run_api:=true VQA_INPUT_IMAGE:=/usb_cam/image_raw gui:=true
... logging to /home/kochigami/.ros/log/aa8bf5b6-9b3b-11ef-a661-4548b1e8015f/roslaunch-kochigami-ThinkPad-P16-Gen-1-236823.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.233.0.246:41955/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /vqa/host: localhost
 * /vqa/image_transport: raw
 * /vqa/port: 8888

NODES
  /
    ofa_api (jsk_perception/run_jsk_vil_api)
    ofa_ros_client_rqt_kochigami_ThinkPad_P16_Gen_1_236823_3430644983751549160 (rqt_reconfigure/rqt_reconfigure)
    ofa_ros_client_rviz_kochigami_ThinkPad_P16_Gen_1_236823_8884148553086753226 (rviz/rviz)
    vqa (jsk_perception/vqa_node.py)

ROS_MASTER_URI=http://localhost:11311

process[ofa_api-1]: started with pid [236861]
process[vqa-2]: started with pid [236886]
process[ofa_ros_client_rviz_kochigami_ThinkPad_P16_Gen_1_236823_8884148553086753226-3]: started with pid [236887]
process[ofa_ros_client_rqt_kochigami_ThinkPad_P16_Gen_1_236823_3430644983751549160-4]: started with pid [236888]
Traceback (most recent call last):
  File "/home/kochigami/catkin_ws/src/jsk_recognition/jsk_perception/node_scripts/vqa_node.py", line 4, in <module>
    from jsk_perception.vil_inference_client import OFAClientNode
ModuleNotFoundError: No module named 'jsk_perception.vil_inference_client'
[vqa-2] process has died [pid 236886, exit code 1, cmd /home/kochigami/catkin_ws/src/jsk_recognition/jsk_perception/node_scripts/vqa_node.py ~image:=/usb_cam/image_raw __name:=vqa __log:=/home/kochigami/.ros/log/aa8bf5b6-9b3b-11ef-a661-4548b1e8015f/vqa-2.log].
log file: /home/kochigami/.ros/log/aa8bf5b6-9b3b-11ef-a661-4548b1e8015f/vqa-2*.log
docker: Error response from daemon: failed to create task for container: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: error running hook #0: error running hook: exit status 1, stdout: , stderr: Auto-detected mode as 'legacy'
nvidia-container-cli: initialization error: nvml error: driver not loaded: unknown.
[ofa_api-1] process has finished cleanly
log file: /home/kochigami/.ros/log/aa8bf5b6-9b3b-11ef-a661-4548b1e8015f/ofa_api-1*.log
```

If anyone has advice on how to solve this, please let me know.

mqcmd196, 2024-11-06:
Thank you for using the feature. I'm investigating now, so please wait for a while

mqcmd196, 2024-11-06:
I followed the current README and built from the source, and confirmed works well.

BTW, 

> Traceback (most recent call last):
>   File "/home/kochigami/catkin_ws/src/jsk_recognition/jsk_perception/node_scripts/vqa_node.py", line 4, in <module>
>     from jsk_perception.vil_inference_client import OFAClientNode
> ModuleNotFoundError: No module named 'jsk_perception.vil_inference_client'

it seems that `jsk_perception` is not built or not in PATH. Could you check the current PATH?

mqcmd196, 2024-11-06:
I also found another issue with your console output.

```
docker: Error response from daemon: failed to create task for container: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: error running hook #0: error running hook: exit status 1, stdout: , stderr: Auto-detected mode as 'legacy'
nvidia-container-cli: initialization error: nvml error: driver not loaded: unknown.
```

It seems that nvidia-container-cli has not been launched correctly. I'm not sure, but it is sometimes fixed by rebooting your PC.

If rebooting does not fix the issue,
1. Please check the nvidia-driver is installed correctly. (We don't need the latest driver. Using `sudo ubuntu-drivers install` is fine if you have NVIDIA GPU )
2. Please check the nvidia-container-toolkit is installed correctly. I followed the official document https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/latest/install-guide.html#installing-with-apt

The simplest test command is  https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/latest/sample-workload.html#running-a-sample-workload-with-docker . If you cannot execute `sudo docker run --rm --runtime=nvidia --gpus all ubuntu nvidia-smi` correctly, there are issues related to GPU drivers or containers.

kochigami, 2024-11-06:
@mqcmd196

Thank you very much for your comment and thank you very much for creating wonderful programs!!
I could execute this!

I found that I couldn’t install nvidia-driver. I also failed to build jsk_perception correctly.
After using `sudo ubuntu-drivers install`, rebooting my PC, and `roscd jsk_perception`  and `catkin bt`, I could execute those commands below.

`nvidia-smi`

```
ipython3
from jsk_perception.vil_inference_client import OFAClientNode
```

I also confirmed that nvidia-container-toolkit is installed correctly and I could execute `sudo docker run --rm --runtime=nvidia --gpus all ubuntu nvidia-smi`

When executing programs, I forgot to launch a server.

<terminal 1>
```
roscore
```

<terminal 2>
```
rosrun usb_cam usb_cam_node
```

<terminal 3>
```
kochigami@kochigami-ThinkPad-P16-Gen-1:~/catkin_ws/src/jsk_recognition/jsk_perception/docker$ ./run_jsk_vil_api --port 8888 --ofa_task caption --ofa_model_scale large ofa
```

<terminal 4>
```
roslaunch jsk_perception  vqa.launch run_api:=true VQA_INPUT_IMAGE:=/usb_cam/image_raw gui:=true
```
[image]

closed 2024-11-06 by kochigami

mqcmd196, 2024-11-06:
Thank you for your report.

> sudo chmod 666 /var/run/docker.sock
 
I think adding yourself to docker group is the best way. `sudo gpasswd -a <username> docker`

kochigami, 2024-11-07:
Thank you very much!! It was helpful!!
(memo for me: I need to reboot my PC after executing the command)