jsk-ros-pkg/jsk_recognition #2865: aws_auto_checkin_appが使えない

sawada10, 2025-04-15:
@k-okada @iory @mqcmd196 
研究会で話に出たaws_auto_checkin_appが使えない件について一旦issueを作ります。

やりたいこと：人の顔と名前を覚えることをしたくて以下を参考にしました。
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/doc/jsk_perception/nodes/aws_auto_checkin_app.rst

**実行手順**
1. サンプルをlaunchする
```
$ roslaunch jsk_perception sample_aws_auto_checkin_app.launch use_aws_face_detection:=true
... logging to /home/leus/.ros/log/f217a19a-19c2-11f0-bcf4-4fc2871093a1/roslaunch-leus-ThinkPad-T480s-110669.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://leus-ThinkPad-T480s:41355/

SUMMARY
========

CLEAR PARAMETERS
 * /aws_auto_checkin_app/
 * /aws_auto_checkin_app/draw_rects/
 * /aws_detect_faces/

PARAMETERS
 * /aws_auto_checkin_app/always_publish: True
 * /aws_auto_checkin_app/approximate_sync: True
 * /aws_auto_checkin_app/aws_credentials_path: /tmp/aws.json
 * /aws_auto_checkin_app/draw_rects/approximate_sync: True
 * /aws_auto_checkin_app/draw_rects/font_path: /opt/ros/noetic/s...
 * /aws_auto_checkin_app/draw_rects/image_transport: compressed
 * /aws_auto_checkin_app/draw_rects/label_size: 16
 * /aws_auto_checkin_app/draw_rects/queue_size: 30
 * /aws_auto_checkin_app/draw_rects/use_classification_result: True
 * /aws_auto_checkin_app/env_path: /tmp/env.json
 * /aws_auto_checkin_app/image_transport: compressed
 * /aws_auto_checkin_app/queue_size: 1
 * /aws_auto_checkin_app/use_window: True
 * /aws_detect_faces/always_publish: True
 * /aws_detect_faces/attributes: ALL
 * /aws_detect_faces/aws_credentials_path: /tmp/aws.json
 * /aws_detect_faces/use_window: True
 * /camera/camera_nodelet_manager/num_worker_threads: 4
 * /camera/depth_metric/queue_size: 5
 * /camera/depth_metric_rect/queue_size: 5
 * /camera/depth_points/queue_size: 5
 * /camera/depth_rectify_depth/interpolation: 0
 * /camera/depth_rectify_depth/queue_size: 5
 * /camera/depth_registered_hw_metric_rect/queue_size: 5
 * /camera/depth_registered_metric/queue_size: 5
 * /camera/depth_registered_rectify_depth/interpolation: 0
 * /camera/depth_registered_rectify_depth/queue_size: 5
 * /camera/points_xyzrgb_hw_registered/queue_size: 5
 * /camera/rgb_rectify_color/queue_size: 5
 * /image_view00/image_transport: compressed
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /use_sim_time: True

NODES
  /
    aws_auto_checkin_app (jsk_perception/aws_auto_checkin_app.py)
    aws_detect_faces (jsk_perception/aws_detect_faces.py)
    camera_base_link (tf2_ros/static_transform_publisher)
    camera_base_link1 (tf2_ros/static_transform_publisher)
    camera_base_link2 (tf2_ros/static_transform_publisher)
    camera_base_link3 (tf2_ros/static_transform_publisher)
    image_view00 (image_view/image_view)
    rosbag_play (rosbag/play)
[... 638 lines cut ...]
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[ INFO] [1744699208.541598178]: Done Setting Dynamic reconfig parameters.
[ INFO] [1744699208.544566490]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1744699208.547738777]: color stream is enabled - width: 848, height: 480, fps: 30, Format: RGB8
[ INFO] [1744699208.552542667]: setupPublishers...
[ INFO] [1744699208.560692326]: Expected frequency for depth = 30.00000
[ INFO] [1744699208.635977171]: Expected frequency for color = 30.00000
[ INFO] [1744699208.707991086]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1744699208.768554377]: setupStreams...
 15/04 15:40:09,169 WARNING [139784350668544] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1744699209.220503826]: SELECTED BASE:Depth, 0
[ INFO] [1744699209.228666503]: RealSense Node Is Up!
[ WARN] [1744699209.692673409]: 
 15/04 15:40:09,693 WARNING [139784350668544] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 15/04 15:40:09,743 WARNING [139784350668544] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
```
rqt_image_viewで`/camera/color/image_raw`からrealsense画像を見れていることを確認できた。
aws関連のtopicも出ているが、rqt_image_viewでそれらを選択してもなにか画像が見れるわけではない。

ここまでまとめると、
- logファイルを入力画像にして動かない
- realsenseを入力画像にして動いているのかわからない（そもそも正しい挙動がよくわかっていない）
という感じです。他の方の環境での挙動や参考になるプログラムなどあれば教えていただけると嬉しいです。

mqcmd196, 2025-04-15:
自分の手元ではサンプルのbagファイルで動作することを確認しました．とりあえず `roslaunch jsk_perception sample_aws_auto_checkin_app.launch use_aws_face_detection:=true --screen` が動く状態を目標にしましょう．

> 上のような実行結果になり、image formatが異なるというエラーが出ているので、logファイルの形式の問題かと考えた。

なんのlogファイルですか？bagファイルの誤り？

予想ですがこれはラベル画像をsubscribeしてしまっているが故に出ているエラーな気がします．ちなみにデフォルトで以下のようにimage_view2が立ち上がるんだけど，これも表示されないかな．

[image]

あと，なんのノードがなんのエラーを出しているかログで詳細に知りたいときは `export ROSCONSOLE_FORMAT='[${severity}][${time}][${node}:${logger}][${file}:L${line}]: ${message}'` みたいに環境変数をセットするとよいです

sawada10, 2025-04-15:
確認ありがとうございます🙇
すみません、logファイルはbagファイルの間違いです

明日は予定があるのですが、明後日もう一度落ち着いて実行してみます

sawada10, 2025-04-17:
先程@k-okada 先生に見ていただいたところ、待ちが足りなかったようで、落ち着いて待ったら動かすことができました。
realsenseでも動かせるようになりました。
ありがとうございました。

k-okada, 2025-04-17:
多分，画面に人の名前が表示されていない気がするんだけど，デフォルトのdeb だとfont_dataがなくて，表示されないです．ということで，
https://github.com/jsk-ros-pkg/jsk_recognition/pulls
のファイルを手動やり `/opt/ros/noetic/share/jsk_perception/node_scripts/`にコピーしたら名前のテキストが出るとおもいます．ひとつだけ注意は，ファイルの１行目の`#!/usr/bin/env python`をpython3に変更しておいて下さい．

@sawada10 の個人環境では，こちらも手動で入れた`jsk_recognition_utils/font_data/` を使うとエラーがでていたので，一旦別の名前にしてください．

sawada10, 2025-04-17:
@k-okada 先生
ありがとうございます。以下をしたら、画面に長方形と名前の両方が表示されることが確認できました。

- https://github.com/jsk-ros-pkg/jsk_recognition/pulls を、手動で`/opt/ros/noetic/share/jsk_perception/node_scripts/draw_rects.py`に反映させる
- jsk_recognition_utils/font_data/以下の`NotoSansJP-Regular.otf`を`NotoSansJP-Regular-backup.otf`に変更する

k-okada, 2025-04-18:
良かったです．
こちらはもう試さなくて良いけど，フォントファイルがあるときに，エラーがでていたのは手元でpipで Pillow(PIL) をインストールしていた，ということで，一応
https://github.com/jsk-ros-pkg/jsk_recognition/pull/2868/commits/5c7612c53294b286fa44e6752e26371c0dcc1afb
で対応しました．

が，もしかしたら将来別のプログラムでまたエラーが出るかもしれません．

pipを使う時は，virtualenv で workspace つくってhttps://qiita.com/ChappyKaori/items/6ccb4fb7065cac26d9b8 その中で作業するのが，他に副作用が無くて安全です．

sawada10, 2025-04-18:
ありがとうございます
覚えておいて、次pipを使う際に参考にします🙇