jsk-ros-pkg/jsk_recognition #2871: detic_rosで/docker/detic_segmentor/output/boxesをpublishする方法が分からない

ishry, 2025-04-28:
JAXONでdetic_rosを使用するために、
https://github.com/HiroIshida/detic_ros?tab=readme-ov-file#detic_ros--
上記リポジトリのReadMeに従って作業していたのですが、
docker/detic_segmentor/output/boxes　というトピックを出す方法について言及がありませんでした。
ご存知の方いらっしゃれば教えていただきたいです。

```
roslaunch detic_ros sample.launch \
    out_debug_img:=true \
    out_debug_segimg:=false \
    compressed:=false \
    device:=auto \
    input_image:=/rs_l515/color/image_rect_color
```

このようにroslaunchしたところ、
```
/docker/detic_segmentor/debug_image
/docker/detic_segmentor/segmentation_info
```
の２つのトピックが出ていることは確認できています。

@mqcmd196

mqcmd196, 2025-04-28:
> docker/detic_segmentor/output/boxes　というトピックを出す

をするためには

https://github.com/HiroIshida/detic_ros#:~:text=Another%20example%20for%20running%20three%20dimensional%20object%20pose%20detection%20using%20point%20cloud%20filtered%20by%20segmentation.

だと思います

mqcmd196, 2025-04-28:
つまり `roslaunch detic_ros sample_detection.launch` のほうです

ishry, 2025-04-29:
`roslaunch detic_ros sample_detection.launch`の実行で、`/docker/detic_segmentor/output/boxes`等様々なトピックが出ることが確認できました。

mqcmd196, 2025-04-29:
Good.

Please reopen if you have other issues.

closed 2025-04-29 by mqcmd196